jbush001/NyuziProcessor #38: Repo split

olofk, 2016-01-29:
I'd say this is more of a suggestion than an issue.

As you already use submodules for the toolchain and verilator, I would suggest you split out the software and RTL code into different repos as well. For the RTL code I would even suggest moving out the board-specific stuff (hardware/fpga/de2_115) and the common peripherals (hardware/fpga/common) so that it will be easier to make create new board ports and reuse your peripheral controllers

jbush001, 2016-01-30:
Yes, I think this is the right approach. At some point, I will probably pull in external projects for things like libc instead of using my own quick and dirty versions, and it will be cleaner to fork separate repos for those. I definitely want to get the FPGA test peripherals and boards out of the tree.

There are a bunch of messy interdependencies between things in the tree now, which I need to work out.

jbush001, 2017-06-23:
At a minimum, the following components could be their own projects:

- tools/emulator
- hardware/core
- hardware/fpga/common (this has a bunch of dependencies on hardware/core). Eventually, would be cleaner to use something like fuseSOC
- tests (should these live with their components?)
- software/libs/librender
- software/apps
- software/kernel (probably replace with a real kernel)
- software/benchmarks

jbush001, 2017-06-24:
Also investigate how to trigger CI build when a subproject is modified.