jbush001/NyuziProcessor #54: Cache performance will degrade in some simulator due to the implement of module "cache_lru"

s117, 2017-05-04:
Recently I have managed to port the Nyuzi to Xilinx ZC706 evaluation board and run doom & quakeview successfully, with respect. But during this process, I noticed a cache performance degradation when debugging with Modelsim SE 10.5c. Since such degradation seems only existing when simulating in some simulators and doesn't exist in verilator or after synthesize, I think maybe this topic is more like a portability suggestion than an issue.

When running the core on some simulators such as Modelsim, the x value coming from uninitialized SRAM will cause the casez statement in module cache_lru fail to match any pattern, further cause that in all cache set only way 0 is available. To illustrate it clearly, here is a piece of code snipping from cache_lru.sv when parameter NUM_WAYS is 4.

```
...

#130            4:
#131            begin
#132                always_comb
#133                begin
#134                    casez (lru_flags)
#135                        3'b00?: fill_way = 0;
#136                        3'b10?: fill_way = 1;
#137                        3'b?10: fill_way = 2;
#138                        3'b?11: fill_way = 3;
#139                        default: fill_way = '0;
#140                    endcase
#141                end
#142
#143                always_comb
#144                begin
#145                    case (new_mru)
#146                        2'd0: update_flags = {2'b11, lru_flags[0]};
#147                        2'd1: update_flags = {2'b01, lru_flags[0]};
#148                        2'd2: update_flags = {lru_flags[2], 2'b01};
#149                        2'd3: update_flags = {lru_flags[2], 2'b00};
#150                        default: update_flags = '0;
#151                    endcase
#152                end
#153            end

...
```

Consider after reset, two or more cache fill operations happened in the same cache set. When the first fill request comes, because the value of "lru_flags" is 3'bxxx (the value of uninitialized SRAM), the default branch will take effect, way 0 got filled, and the "update_flags" will be 3'b11x. Then, in the subsequent request handling process, the casez statement will always fail to match any pattern with the value of "lru_flags" (3'b11x), the default branch will always take effect, way 0 will always get filled, and "update_flags" will always be 3'b11x. The similar issue also exists when NUM_WAY is 8. The marked red area in the waveform generated by Modelsim attached below shows such situation:

[image]

Although it sounds like a bad idea, one solution to this problem is changing all casez in cache_lru.sv to casex. I made an experiment to estimate the performance impact: Some logic was inserted into the writeback_stage firstly to trace the function activity. Then, the same program got executed on two different designs, the original design and the design with modified cache_lru, for a given time. Since the program is same, the amount of log generated during simulation is related to the instructions got executed. By comparing the number of function tracing record, we can compare the performance approximately. The screenshot below shows the result: The design with modified cache_lru produced 2.8x more log than the original one, which can roughly be considered as running faster 2.8x.

[image]

The version I use is ce221ff, but this issue still exists in the latest version.

Finally, Thanks for bringing such interesting project!

[image]

jbush001, 2017-05-05:
Good catch! Nice write up of the issue too, thanks.

Since Verilator is only a two-state simulator (no x or z), it doesn't manifest this problem, but any four-state simulator probably will. Another solution might be to initialize the SRAMs to 0 at startup.

jbush001, 2017-05-05:
This patch should address the issue, but feel free to let me know if not.

s117, 2017-05-05:
Sadly, if a logic/reg variable is driven in an always_ff block, the "vlog" compiler in the latest Modelsim seems doesn't allow any other block drive that single again, even it's an initial block. So the patch 7780328 just can't pass the compile. 

[image]

The following patch is ok with the help of macro SIMULATION, but not very beautiful...

```
--- sram_1r1w_ce221ff.sv        2017-03-23 23:05:54.677195800 +0800
+++ sram_1r1w_patched.sv        2017-05-05 16:32:46.063266600 +0800
@@ -100,7 +100,11 @@
     // Simulation
     logic[DATA_WIDTH - 1:0] data[SIZE];

+`ifdef SIMULATION
+    always @(posedge clk)
+`else
     always_ff @(posedge clk)
+`endif
     begin
         if (write_en)
             data[write_addr] <= write_data;
@@ -120,6 +124,11 @@

     initial
     begin
+`ifdef SIMULATION
+        for (int i = 0; i < SIZE; i++)
+            data[i] = '0;
+`endif
+
         if ($test$plusargs("dumpmems") != 0)
             $display("sram1r1w %d %d", DATA_WIDTH, SIZE);
     end
```

```
--- sram_2r1w_ce221ff.sv        2017-03-22 23:48:00.614532400 +0800
+++ sram_2r1w_patched.sv        2017-05-05 16:32:48.783422200 +0800
@@ -128,7 +128,11 @@
     // Simulation
     logic[DATA_WIDTH - 1:0] data[SIZE];

+`ifdef SIMULATION
+    always @(posedge clk)
+`else
     always_ff @(posedge clk)
+`endif
     begin
         if (write_en)
             data[write_addr] <= write_data;
@@ -160,6 +164,11 @@

     initial
     begin
+`ifdef SIMULATION
+        for (int i = 0; i < SIZE; i++)
+            data[i] = '0;
+`endif
+
         if ($test$plusargs("dumpmems") != 0)
             $display("sram2r1w %d %d", DATA_WIDTH, SIZE);
     end
```

jbush001, 2017-05-05:
Yeah, a bit hacky, but I'm not sure of another way to do this cleanly now. Pushed another patch.

I didn't add the SIMULATION guards in this patch. Did you need that because you are inferring the SRAM for the Xilinx part in that block? I would have expected the last block to not infer SRAMs properly (with an area and clock speed hit), but perhaps this works okay on Xilinx.  It definitely doesn't on Altera. I *think* it might be preferable to have a VENDOR_XILINX block and explicitly instantiate the block RAM, like how it is done in VENDOR_ALTERA, to avoid any issues.

s117, 2017-05-05:
Problem solved with 77a50a8, thanks for the time!

Yes, currently the SRAM in my design is inferred from this block because it just works... Thanks for the advice. I will make a VENDOR_XILINX block and submit another pull request after finished.

closed 2017-05-05 by s117