jbush001/NyuziProcessor #57: Floating point multiplication exponent underflow handled incorrectly

jbush001, 2017-05-28:
The following code:

```c
float a = 3.40282347E-24F;
float b = 3.40282347E-24F;

int main()
{
    union {
        float fval;
        int ival;
    } u;
    
    u.fval = a * b;
    printf("%08x\n", u.ival);
}
```

Should print 00000000, but prints 7187625d when run in Verilog simulation. Hardware needs to explicitly detect if the sum of the exponents underflows and set the result to zero.

jbush001, 2017-09-02:
https://github.com/jbush001/NyuziProcessor/blob/0abeff4141498d58524fb742c454572bc1d23d54/tests/core/isa/float.S#L134

manili, 2017-11-16:
I think this issue has been resolved with commit https://github.com/jbush001/NyuziProcessor/commit/cd7306aeb7dc44b07238f8ba07b3d6daf67e5c7f and now it can be closed.

closed 2017-11-19 by jbush001