jbush001/NyuziProcessor #106: Error configuring toolchain

kunsa, 2017-07-06:
could you tell the error how to be sovled or it don't need to be solved? THx!
first: git clone https://github.com/jbush001/NyuziProcessor.git
second: cd NyuziProcessor/
third: build/setup_tools.sh
it comes:
CMake Error at tools/lldb/unittests/tools/lldb-server/CMakeLists.txt:10 (add_definitions):
  Error evaluating generator expression:

    $<TARGET_FILE:lldb-server>

  No target "lldb-server"

CMake Error at tools/lldb/unittests/tools/lldb-server/CMakeLists.txt:10 (add_definitions):
  Error evaluating generator expression:

    $<TARGET_FILE:lldb-server>

  No target "lldb-server"

CMake Error at tools/lldb/unittests/tools/lldb-server/CMakeLists.txt:10 (add_definitions):
  Error evaluating generator expression:

    $<TARGET_FILE:lldb-server>

  No target "lldb-server"

CMake Error at tools/lldb/unittests/tools/lldb-server/CMakeLists.txt:10 (add_definitions):
  Error evaluating generator expression:

    $<TARGET_FILE:lldb-server>

  No target "lldb-server"

-- Generating done
-- Build files have been written to: /home/user/NyuziProcessor/tools/NyuziToolchain/build
Error configuring toolchain

kunsa, 2017-07-06:
and then : make
it comes : 
Building dependencies for src/memcmp.c
make[3]: /usr/local/llvm-nyuzi/bin/clang: Command not found
Building dependencies for src/memset.c
make[3]: /usr/local/llvm-nyuzi/bin/clang: Command not found
Building dependencies for src/vfprintf.c
make[3]: /usr/local/llvm-nyuzi/bin/clang: Command not found
Building dependencies for src/time.c
make[3]: /usr/local/llvm-nyuzi/bin/clang: Command not found
Building dependencies for src/qsort.c
make[3]: /usr/local/llvm-nyuzi/bin/clang: Command not found
Building dependencies for src/stdlib.c
make[3]: /usr/local/llvm-nyuzi/bin/clang: Command not found
Building dependencies for src/cxx_runtime.cpp
make[3]: /usr/local/llvm-nyuzi/bin/clang: Command not found
Building dependencies for src/dlmalloc.c
make[3]: /usr/local/llvm-nyuzi/bin/clang: Command not found
Building dependencies for src/string.c
make[3]: /usr/local/llvm-nyuzi/bin/clang: Command not found
Building dependencies for src/stdio.c
make[3]: /usr/local/llvm-nyuzi/bin/clang: Command not found
Building dependencies for src/math.c
make[3]: /usr/local/llvm-nyuzi/bin/clang: Command not found
mkdir -p obj
Compiling src/math.c
make[3]: /usr/local/llvm-nyuzi/bin/clang: Command not found
../../..//build/target.mk:57: recipe for target 'obj/math.o' failed
make[3]: *** [obj/math.o] Error 127
make[3]: Leaving directory '/home/user/NyuziProcessor/software/libs/libc'
Makefile:18: recipe for target 'all' failed
make[2]: *** [all] Error 2
make[2]: Leaving directory '/home/user/NyuziProcessor/software/libs'
Makefile:18: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Leaving directory '/home/user/NyuziProcessor/software'
Makefile:18: recipe for target 'all' failed
make: *** [all] Error 2

jbush001, 2017-07-06:
Oops, I had fixed this in the toolchain branch but forgot to update the subproject. I just did that. Running the following commands should fix it:

```
git submodule update
rm -f tools/NyuziToolchain/Makefile
./tools/setup_tools.sh
```

Let me know how that goes.

kunsa, 2017-07-06:
THx! Just know that you are doing this interesting thing, though know little, but very interested in it！Iwill often consult you about it in the future ! Thx again.

kunsa, 2017-07-06:
I'm sorry to tell you that there's nothing in NyuziProcessor/tools/NyuziToolchain and setup_tools.sh isn't in NyuziProcessor/tools as well , so ./tools/setup_tools.sh doesn't work /(ㄒoㄒ)/~~

jbush001, 2017-07-06:
Huh, the directories are empty?  I'd suggest deleting the whole project on
your machine and cloning again.

jbush001, 2017-07-07:
**(I mistyped the second path, which should have been ./build/setup_tools.sh, but at any rate, just blowing away the repo and resyncing it is simpler and will get you all of the latest stuff).**

kunsa, 2017-07-08:
It is effective to solve the error, but "setup_tools.sh" spends too much time .  And I will go on...

jbush001, 2017-07-08:
LLVM is big, it takes a while to build. :)

jbush001, 2017-07-08:
I'll close this for now, since the original issue seems to be resolved. Feel free to let me know if you run into any other problems.

closed 2017-07-08 by jbush001