jbush001/NyuziProcessor #107: learn skills

kunsa, 2017-07-08:
Could you give me some advises or guidance to learn about Nyuzi ? THx

jbush001, 2017-07-08:
A good starting point would be the wiki section of this project, which has descriptions of the programming model:
[https://github.com/jbush001/NyuziProcessor/wiki/Instruction-Set](https://github.com/jbush001/NyuziProcessor/wiki/Instruction-Set)

And hardware implementation:
[https://github.com/jbush001/NyuziProcessor/wiki/Microarchitecture](https://github.com/jbush001/NyuziProcessor/wiki/Microarchitecture)

These assume some background understanding of processor design. Here are some resources, from basic to more complex:

This is a quick intro to the classic RISC pipeline, which you should understand thoroughly (the Nyuzi pipeline is much more complex, but builds on some of the basic concepts): https://en.wikipedia.org/wiki/Classic_RISC_pipeline

A good introduction to processor design is general is the classic [Computer Architecture a Quantitative Approach](https://www.amazon.com/Computer-Architecture-Quantitative-Approach-book/dp/1558603727/ref=sr_1_6?ie=UTF8&qid=1499533262&sr=8-6&keywords=computer+architecture+a+quantitative+approach).There are often fairly cheap used copies available.  **I would recommend getting the 1st or 2nd edition and not a newer one.** Unfortunately, the newer editions removed some of the best material and pushed the rest to an appendix.

It might also be useful to look at a few simpler processor variants to understand how these concepts are applied. The LM32 processor is pretty clean:

http://www.latticesemi.com/~/media/LatticeSemi/Documents/UserManuals/JL/LatticeMico32ProcessorReferenceManual39.pdf?document_id=52077

The VScale processor is a simple pipelined implementation supporting the RISC-V instruction set:

https://github.com/ucb-bar/vscale

closed 2017-07-08 by jbush001

jbush001, 2017-07-20:
Also check out "[Computer Organization and Design](https://www.amazon.com/Computer-Organization-Design-Fourth-Architecture/dp/0123747503/ref=sr_1_4?s=books&ie=UTF8&qid=1500557875&sr=1-4&keywords=computer+organization+and+design)" by the same authors, which describes how instruction sets work and also gives a description of pipelining and hazards in chapter 4.

jbush001, 2017-07-20:
I've heard this online course, which has an associated textbook, is really good: [http://nand2tetris.org/](http://nand2tetris.org/) but I don't have any experience with it.

kunsa, 2017-07-22:
Learning plan and progress：
1.Computer Architecture A Quantitative Approach(third edition)
1st chapter Fundamentals of Computer Design(done)
2nd chapter Instruction Set Principles and Examples(done)
3rd chapter Instruction-Level Prallelism and its Dynamic Exploitation(done)
2.Appendix A Pipeling: Basic and Intermediate Concepts(done)
3.Computer Organization and Design(forth edition)
4th chapter The Processor (going on)
Does it an appropriate plan?

jbush001, 2017-07-22:
I think that should give you a good background. Is it making sense to you?

kunsa, 2017-07-22:
Yes, it tastes good^_^, and I find the text for the online course: ( http://nand2tetris.org/)Building a Modern Computer From First Principles , and other texts are great, but I don't view them all.
http://phoenix.goucher.edu/~kelliher/   --> CS 220: Computer Architecture.

kunsa, 2017-08-11:
Hi! 
I have learned the 1st-6th chapters (Boolean Logic, Boolean Arithmetic, Sequantial Logic, Machine Language, Computer Architecture, Assembler) of the book "the elements of computing systems" from the online course, and the following chapters is about the structures of "the computer: Jack", so I turn to your recommended books : Computer Organization and Design.  Its name is "The Processor" in  chapter 4 which describes the design of instruction sets and gives a description of pipelining and hazards.  Do you have some advises about what should I do next? Thx!

jbush001, 2017-08-11:
I don't have any more specific recommendations. Is the wiki description of the architecture starting to make sense? If not, what parts are still unclear?

kunsa, 2017-08-12:
I have clear understanding for  the wiki description of Classic RISC pipeline after reading Computer Organization and Design (4th).  Had I better start from the code? or some other information from other books?

jbush001, 2017-08-12:
I'd suggest looking at the code.

kunsa, 2017-08-13:
OK, I will start from NyuziProcessor/software/kernel/ , and is it a suitable way to debug it by the gdb tool ?

jbush001, 2017-08-13:
That doesn't seem like the best place to start if you're interested in learning about the hardware. I would suggest looking at hardware/core.

kunsa, 2017-08-17:
Could you teach me a method to understand the code in  hardware/core. I roughly looked at the whole folder, but cannot understand how to link together. Is it a Verilog HDL language? And where is the "main" funtion?

kunsa, 2017-08-31:
I have got the point of Verilog HDL, and read two books about it. When having an overview of the entire code and read “hardware/core/config.sv,core.sv,defines.sv,nyuzi”, I find it hard to understand the whole process and program architecture, could you teach me an easy way to read it clearly ? Thx!

jbush001, 2017-09-01:
The execution pipeline is defined here: https://github.com/jbush001/NyuziProcessor/blob/1c461e8d0922e70a3c3ea9d3e11657946f6041a0/hardware/core/core.sv#L332-L345

One approach might be to try walking through the path an instruction takes through the pipeline, reading the description of each pipeline stage here: https://github.com/jbush001/NyuziProcessor/wiki/Microarchitecture#execution-pipeline and then looking at the corresponding verilog module to see how those ideas are implemented.  For example, the section of the wiki page for instruction cache tag describes this:

"This stage has a program counter for each hardware thread. It selects one to fetch each cycle,"

Looking at hardware/core/ifetch_tag_stage.sv, I see this:

```
    assign pc_to_fetch = next_program_counter[selected_thread_idx];
```

and so on...

kunsa, 2017-09-13:
I have read all the code in hardware/core, and some details are not fully understood. Could you tell me how to  emulate it in software? Thx!

jbush001, 2017-09-14:
Are you asking how to run the hardware design in verilog simulation? If so, try going into software/apps/hello and typing 'make verirun'. You can set the variable DUMP_WAVEFORM when building the hardware design to output a waveform trace file. See instructions in the readme [here](https://github.com/jbush001/NyuziProcessor/blob/6134b02243da2d7976ecb107363244c4eb47fb99/hardware/README.md).

kunsa, 2017-09-16:
Yes. If it means that: 1.git clone in linux 2.enter software/apps/hello 3. typing 'make verirun' 4.make clean
DUMP_WAVEFORM=1 make 
like this way?

jbush001, 2017-09-16:
You'd need to follow the instructions in the top level README to set up the project first. Simulation in Verilog does not require an FPGA.

kunsa, 2017-09-16:
OK! I'm tring~