jbush001/NyuziProcessor #140: SD card reads fail on FPGA

jbush001, 2018-01-29:
Times out initializing card. It seems to never read anything but 0xff.

jbush001, 2018-02-04:
1. Connected pins to GPIO block. I verified I could read the input pins (printed to the serial log) by shorting the solder pads to ground, and drive output pins by controlling with software and measuring with a multimeter.
2. I tried putting the chip into 4 bit mode. I could see correct data on DAT1-3, but DAT0 was stuck at VCC. I wrote various patterns on the SD card on my laptop, then re-read on the board to confirm.

jbush001, 2018-02-24:
Next steps:
- I'd like to obtain a USB logic analyzer/oscilloscope and probe what the signals are actually doing. Check both the timing of logic signals, but also analog characteristics like voltage to ensure they are in spec.
- One theory I have is that there is a mechanical problem with the connector, which seems like wishful thinking, but I could verify that by attaching an external SD card connector to one of the other headers.

jbush001, 2018-02-26:
Simulation waveforms. Here is the initial clock sequence:

[image]

Here is CMD0:

[image]

```
01000000 00000000 00000000 00000000 00000000 10010101
40       00       00       00       00       95
CMD0 
``

jbush001, 2018-02-26:
Verified 250 system clocks per spi clock: 200kHz SPI clock at 50Mhz system clock.

jbush001, 2018-02-26:
While waiting for response:

[image]

(Should CS be high here?)

jbush001, 2018-02-26:
Tried with 3 cards, all failed in the same way (no response at all)
- 8 GB Samsung microSD (in adapter)
- 4 GB Monster SDHC
- 8 MB Canon

jbush001, 2018-03-03:
I hooked up a logic analyzer and the first command worked correctly.

[image]

Unplugged the logic analyzer and it started failing again... Seems like this might be a connector issue.

jbush001, 2018-03-03:
I also had to turn the board upside down to attach the probes. In the normal position, gravity pulls the card down away from the pins. Upside down, it pushes them against them. I might try sticking a piece of paper in with the card to apply more pressure.

jbush001, 2018-03-03:
I tried sticking paper in, turning the board around with no success. However, it works consistently if the probes are attached...

[image]

closed 2018-03-03 by jbush001

travisg, 2018-03-03:
Hmm, what board is that BTW?

jbush001, 2018-03-03:
DE2-115

travisg, 2018-03-03:
Ah didn't know if you were still using that. I actually bought one of those for myself when this project got started up years ago. I should brush it off.