jbush001/NyuziProcessor #145: Improve instruction encoding

jbush001, 2018-02-10:
The current encoding leaves gaps of unused encoding space. Unlike other architectures, the instruction format is explicitly encoded. For each format, there are additional fields that specify subformats. For example, a memory instruction begins with the prefix '10'. There is then another 4-bit operation field. However, there are only 11 memory operations types, so there are ~335 million encodings that are unused.

This encoding scheme was originally done to simplify decoding. However, as I've needed to add instructions, I've started stuffing them into forms that don't make a lot of sense, which is making it much more complex than other encoding schemes.

A better approach might be to use a fixed width opcode field, with the rest of the format implied from the opcode, similar to many other RISC processors like RISC-V and Mips.

jbush001, 2018-02-11:
[image]

jbush001, 2018-02-11:
opcode | description | format
-- | -- | --
0 | break | n/a
1 | int arithmetic SS | R
2 | int arithmetic VS | R
3 | int arithmetic VV | R
4 | int arithmetic VSM | R
5 | int arithmetic VVM | R
6 | int arithmetic SI | I
7 | int arithmetic VI | I
8 | int arithmetic VIM | IM
12 | unary scalar | R
13 | unary vector | R
14 | compare int VV | R
15 | compare int VI | I
16 | compare int VS | R
17 | compare int SI | I
18 | compare float VV | R
19 | compare float VS | R
20 | compare float SS | R
21 | syscall | R
22 | LUI | J
23 | AUIPC | J
24 | shuffle | R
25 | getlane reg | R
26 | getlane imm | I
27 | load | I
28 | store | I
29 | load masked | IM
30 | store masked | IM
31 | getcr | I
32 | setcr | I
33 | cache control | I
34 | call | LJ
35 | call register | J
26 | bzero | J
27 | bnzero | J
28 | b (reg) | J
29 | b (offset) | J
30 | eret | n/a

jbush001, 2018-02-11:
[image]

jbush001, 2018-02-11:
[image]

jbush001, 2018-02-11:
This is heavily inspired by RISC-V's encoding, but adapted to work with Nyuzi's ISA.

This leaves a lot more instruction encoding space open, at the expense of some immediate range (12 bits vs. 14). The control register transfers should use the immediate field, extending the number from 32 to 4096, which will give much more room for expansion. Having a fixed opcode field should simplify decoding. All of the registers are in the same locations in the instruction, reducing the number of muxes and decoder delay.

jbush001, 2018-02-11:
This allows solving issue #146 and #135 more cleanly, without having to bastardize the arithmetic formats.

jbush001, 2018-02-17:
- [ ] in compiler Modify lib/Targets/Nyuzi/NyuziInstrFormats.td and NyuziInstrInfo.td to support new formats
- [ ] Modify other code in backend that makes format assumptions (fixup code, etc)
- [ ] Update tests/MC/Nyuzi for new format. Rewrite make_tests.py
- [ ] Modify lld so fixups know new locations 
- [ ] Modify elf2bin to use new jump format
- [ ] Modify LLDB for break instruction
- [ ] Modify emulator to support new format
- [ ] Modify instruction decode stage

jbush001, 2018-02-17:
[image]

[image]

## Integer Arithmetic

| Op | Code |
|----|----|
|or | 0 |
|and | 1 |
|xor | 2 |
|add_i | 3 |
|sub_i | 4 |
|mull_i | 5 |
|mulh_i | 6 |
|mulh_u | 7 |
|ashr | 8 |
|shr | 9 |
|shl | 10 |

XXX Use _u and _i suffixes for shifts

## Integer Comparison

| Op | Code |
|----|----|
| cmpeq_i  | 0 |
| cmpne_i  | 1 |
| cmpgt_i  | 2 |
| cmpge_i  | 3 |
| cmplt_i  | 4 |
| cmple_i  | 5 |
| cmpgt_u  | 6 |
| cmpge_u  | 7 |
| cmplt_u  | 8 |
| cmple_u  | 9 |

## Floating Point Arithmetic

| Op | Code |
|----|----|
| add_f | 0 |
| sub_f | 1 |
| mul_f | 2 |

## Floating Point Comparison

| Op | Code |
|----|----|
| cmpeq_f | 0 |
| cmpne_f | 1 |
| cmpgt_f | 2 |
| cmpge_f | 3 |
| cmplt_f | 4 |
| cmple_f | 5 |

## Unary Arithmetic
| Op | Code |
|----|----|
| ftoi | 0 |
| itof | 1 |
| reciprocal | 2 |
| sext8 | 3 |
| sext16 | 4 |
| ctz | 5 |
| clz | 6 |

## Memory Access
| Op | Code |
|----|----|
| load/store u8 | 0 |
| load s8 | 1 |
| load/store u16 | 2 |
| load s16 | 3 |
| load/store 32 | 4 |
| loadv/storev | 5 |
| load scat/gath | 6 |
| load/store sync | 7 |

## Cache Control
| Op | Code |
|----|----|
| itlbinsert | 0 |
| dtlbinsert | 1 |
| tlbinval | 2 |
| tlbinvalall | 3 |
| dinvalidate | 4 |
| iinvalidate | 5 |
| dflush | 6 |
| membar | 7 |

travisg, 2018-02-18:
Oh that's a lovely set of tables. What tool are you using to generate them?

jbush001, 2018-02-18:
The bytefield package in Latex:

```Latex
\usepackage{bytefield}

\begin{bytefield}[rightcurly=.,endianness=big,bitwidth=1.25em]{32}
  \bitheader{0-31} \\
  \begin{rightwordgroup} {R}
	& \bitbox{6}{opcode (6)}
	& \bitbox{5}{src1 (5)}
	& \bitbox{5}{dest (5)}
	& \bitbox{6}{func (6)}
	& \bitbox{5}{src2 (5)}
	& \bitbox{5}{mask (5)}
  \end{rightwordgroup}\\

  \begin{rightwordgroup} {I}
	\bitbox{6}{opcode (6)}
	& \bitbox{5}{src1 (5)}
	& \bitbox{5}{dest (5)}
	& \bitbox{4}{func (4)}
	& \bitbox{12}{imm (12)}
  \end{rightwordgroup}\\
\end{bytefield} \\
```

jbush001, 2018-02-18:
http://texdoc.net/texmf-dist/doc/latex/bytefield/bytefield.pdf

travisg, 2018-02-18:
Very cool, thanks!

jbush001, 2018-08-01:
An alternative would be to base this on RISC-V, using the RV32I with M+A+F extensions, then adding a non-standard extension for the vector operations.

jbush001, 2019-05-13:
The official RISC-V 'V' vector extensions seems to support most of the vectoroperations needed:

https://riscv.github.io/documents/riscv-v-spec/riscv-v-spec.pdf