jbush001/NyuziProcessor #150: UART doesn't have flow control

jbush001, 2018-03-04:
RTS/CTS are not connected. This doesn't seem to have been a problem so far, since most USB->Serial converts have a large buffer and USB is faster than serial. On the host->device direction, the software download cases read data in a tight loop, so it's not really possible to overrun the FIFO.

jbush001, 2018-03-17:
RTS(RTR) is active low. This is fairly easy to implement, but would require a bit more work to test in simulation:

1. In uart.sv, when reading STATUS_REG, tx_ready is used to indicate the sender can transmit. This should be && !rts
2. The CTS output should be connected to rx_fifo_full.

travisg, 2018-03-17:
Also remember that bug that <vendor> had where it would stop transmitting in the middle of the character if the receiver dropped CTS in the middle of it? IIRC the spec says that you sample the CTS of the other side before starting the character, but then commit to sending it.

jbush001, 2018-03-17:
Haha... yes, I remember that all too well. :) This proposal actually doesn't control transmission, it just sets the status bit to indicate it's not ready. If software ignored that, it could start transmission anyway. But once it has started transmitting a character, it always finishes before checking for the next (controlled by the shift_count variable):

https://github.com/jbush001/NyuziProcessor/blob/beb8c7f6233da1a38da1b5c47fb6fbc6e13f6207/hardware/fpga/common/uart_transmit.sv#L39-L74

I'm not especially proud of this implementation; it's pretty quick and dirty. However, I've seen the RTL for some production chips and it's even worse. :)