jbush001/NyuziProcessor #168: Hardcoded installation path of verilator in unit tests

cpehle, 2018-09-19:
Currently verilator installation path is hardcoded in "NyuziProcessor/tests/unit/runtest.py", this prevents use of non-standard installation paths (for example if the user does not have root permissions or if verilator is installed as a module). 

https://github.com/jbush001/NyuziProcessor/blob/270064eed70cf4174b40f5d287aac28ba606934b/tests/unit/runtest.py#L106-L116

jbush001, 2018-09-20:
Okay, that breaks in continuous integration, because it's not picking up /usr/local/bin/ from the PATH. 

On my machine, VERILATOR_ROOT and VERILATOR_BIN are not set on install (these appear to be build variables).

jbush001, 2018-09-20:
I kind of mangled the commit log, but /usr/local/bin/verilator is a perl script (which then invokes the actual executable), so it can't be invoked directly by subprocess.call. My original workaround was to execute 'sh' with verilator as the argument. But that doesn't perform a PATH lookup (since it's an argument), hence the hardcoded path. Instead, invoking subprocess.call with shell=true uses PATH as expected.

closed 2018-09-20 by jbush001