jbush001/NyuziProcessor #184: CMake Error: CMake was unable to find a build program corresponding to "Ninja".

schuler1, 2019-07-07:
I am running Ubuntu 18.04 and getting the following error when running ./scripts/setup_tools.sh:

```
CMake Error: CMake was unable to find a build program corresponding to "Ninja".  CMAKE_MAKE_PROGRAM is not set.  You probably need to select a different build tool.
CMake Error: CMAKE_C_COMPILER not set, after EnableLanguage
CMake Error: CMAKE_CXX_COMPILER not set, after EnableLanguage
CMake Error: CMAKE_ASM_COMPILER not set, after EnableLanguage
-- Configuring incomplete, errors occurred!
```

Here are a few lines before the previous error for reference purposes:

```
Installed binaries to /usr/local/bin/verilator
Installed man to /usr/local/share/man/man1
Installed examples to /usr/local/share/verilator/examples
```

Is there an additional package that I should download?

schuler1, 2019-07-07:
Perhaps this solves the problem: sudo apt-get install ninja-build
It probably needs to be added to the README file.

jbush001, 2019-07-07:
Oops, yes, you are right. I will update the file. Thanks!

closed 2019-07-14 by jbush001