jbush001/NyuziProcessor #192: I want to use quartus to generate the RTL of GPU

CNJasonChio, 2020-03-18:
I imported files in hardware/core/ into the project, and the following error occurred during the compilation process:
1. Error (10174): Verilog HDL Unsupported Feature error at sram_1r1w.sv(204): system function "$random" is not supported for synthesis
2. Error (10174): Verilog HDL Unsupported Feature error at sram_1r1w.sv(208): system function "$test$plusargs" is not supported for synthesis
Is there any way to solve these errors or generate RTL

GuillemCabo, 2020-03-18:
I'm not directly involved with the project but this is my advice. 

$random is a system function for simulation only and  $test$plusargs looks like system function/call for simulation as well. If you are doing synthesis both calls will have no effect, that is normal.

Double check that you are not loading unnecessary simulation files in to your synthesis run. If the files are necessary then you can guard the problematic lines with ``` `ifdef simulation````or neutralize such errors within quartus.

jbush001, 2020-03-18:
You need to define VENDOR_ALTERA in your project so it uses Altera specific block ram IP.  Go to the Assignments menu and select the 'settings' entry.  Click on the 'Verilog HDL' entry in the list on the left side of the window. In the Name next box, type VENDOR_ALTERA and in the value text box put 1. Then click the 'add' button.

[image]

Note that there is a Quartus project file already in [hardware/fpga/de2-115/](https://github.com/jbush001/NyuziProcessor/tree/master/hardware/fpga/de2-115). I assume you created your own project because you're targeting a different chip, but, if not, you can just use that one.

jbush001, 2020-03-28:
I'm going to close this now, but let me know if you have any issues.

closed 2020-03-28 by jbush001