jbush001/NyuziProcessor #193: Error building NyuziProcessor (probably NyuziToolchain)

gauravjain14, 2020-03-28:
Hi All,
I am trying to build on Ubuntu 16.04 with Python 3.7 and when I run the command `./scripts/setup,sh`, I get the following error after it asks for `sudo `password the first time. 

```
-- Native target X86 is not selected; lli will not JIT code
-- Threads enabled.
-- Doxygen disabled.
-- Go bindings disabled.
-- OCaml bindings disabled, need ctypes >=0.4.
-- Could NOT find Python module pygments
-- Could NOT find Python module pygments.lexers.c_cpp
-- Could NOT find Python module yaml
-- LLVM host triple: x86_64-unknown-linux-gnu
-- LLVM default target triple: nyuzi-none-none
-- Building with -fPIC
-- Constructing LLVMBuild project information
-- Linker detection: GNU ld
-- Targeting Nyuzi
-- Clang version: 9.0.0
-- LLD version: 9.0.0
-- LLDB version: 9.0.0svn
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE)
-- Symbols (liblldb): exporting all symbols from the lldb namespace
-- Failed to find LLVM FileCheck
-- git Version: v0.0.0
-- Version: 0.0.0
-- Performing Test HAVE_GNU_POSIX_REGEX -- failed to compile
-- Performing Test HAVE_POSIX_REGEX -- success
-- Performing Test HAVE_STEADY_CLOCK -- success
-- Configuring done
-- Generating done
-- Build files have been written to: NyuziProcessor/tools/NyuziToolchain/build
[0/4] Performing build step for 'builtins'
ninja: no work to do.
[1/4] No install step for 'builtins'
[4/4] Completed 'builtins'
log: reading configuration file: install
die: error: unable to read configuration file
Error installing toolchain
```

Has anyone seen this before? This comes up after verilator has been installed and it is trying to install NyuziToolChain.

jbush001, 2020-03-28:
This doesn't look familiar. I will do a clean install on my setup to try to reproduce it.

gauravjain14, 2020-03-28:
Thanks for responding quickly. 

That'd be great. I'm not sure if I am missing any dependencies or is it related to permissions because the error msg didn't seem to say anything. 

I'll wait your response. Thanks.

jbush001, 2020-03-28:
Unfortunately, I was not able to reproduce on my setup:

```
[3644/3644] Linking CXX static library lib/libbenchmark_main.a
[sudo] password for jeffbush: 
[2/5] Performing build step for 'builtins'
ninja: no work to do.
[3/5] No install step for 'builtins'
[5/5] Install the project...
-- Install configuration: "Release"
-- Installing: /usr/local/llvm-nyuzi/lib/libLTO.so.9svn
-- Up-to-date: /usr/local/llvm-nyuzi/lib/libLTO.so
```

Configuration:

    Ubuntu 16.04 LTS
    ninja 1.5.5
    cmake 3.5.1
    python 3.5.2

Yeah, that error message is not very helpful. :) I assume it is coming from ninja, but it's unclear. I will do a little more research to figure out where it is originating from. Could you please try the following?

    cd tools/NyuziToolchain/build
    ninja -v -d explain

gauravjain14, 2020-03-29:
So I remove the directory and tried running `./scripts/setup.sh`, but I am getting the same error. 

My Configuraion:
```
Ubuntu 4.15.0-88-generic #88~16.04.1-Ubuntu SMP Wed Feb 12 04:19:15 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
ninja 1.9.0
cmake 3.14.0
python 3.7.3
```

Output of 
```
ninja -v -d explain

ninja explain: output tools/lldb/CMakeFiles/finish_swig doesn't exist
ninja explain: tools/lldb/CMakeFiles/finish_swig is dirty
ninja explain: tools/lldb/finish_swig is dirty
ninja explain: output runtimes/builtins-stamps/builtins-build doesn't exist
ninja explain: runtimes/builtins-stamps/builtins-build is dirty
ninja explain: runtimes/builtins-stamps/builtins-install is dirty
ninja explain: runtimes/builtins-stamps/builtins-build is dirty
ninja explain: runtimes/builtins-stamps/builtins-install is dirty
ninja explain: runtimes/CMakeFiles/builtins-complete is dirty
ninja explain: runtimes/CMakeFiles/builtins is dirty
ninja explain: runtimes/CMakeFiles/builtins-complete is dirty
ninja explain: runtimes/builtins-stamps/builtins-done is dirty
ninja explain: runtimes/builtins-stamps/builtins-install is dirty
ninja explain: runtimes/builtins-stamps/builtins-build is dirty
ninja explain: runtimes/builtins is dirty
```

Does this give any idea? 

I also have a log file for the build. I could you that directly if that helps.
Thanks

jbush001, 2020-03-29:
I think the full log would be helpful (maybe create a gist and add a link to it). Thanks!

gauravjain14, 2020-03-29:
Here's the link to the gist. It's quite long (that's expected) but I think that the last 50 lines would be of relevance to you. 

https://gist.github.com/gauravjain14/10691e9abf6cde6bc12978ce7000d140

Thanks

jbush001, 2020-03-29:
One more thing, can you try from the tools/NyuziToolchain/build directory?

    ninja -v install

jbush001, 2020-03-29:
Sorry, should be:

    sudo ninja -v install

gauravjain14, 2020-03-29:
Hi Jeff,
I think we are missing something because I tried the same `./scripts/setup.sh` on a different machine and saw the build process fail at the same point. 

Then on both the machines I cd'd into `tools/NyuziToolchain/build` and ran the command 

`sudo ninja -v install`

and on both the machines I get the same error

```
log: ninja version 0.1.3 initializing
log: magic group: gid=0 (root)
log: entering main loop
log: generating initial pid array..
log: now monitoring process activity
die: got signal SIGSEGV -- segmentation fault
```

Both are Ubuntu 16.04, with Python 3.7+

jbush001, 2020-03-29:
Thank you, that is really helpful.  I will try upgrading my software, as it appears to be specific to newer versions of the tools. It looks like it's actually crashing ninja. A workaround you could try is using make instead of ninja. Here is a modified version of setup.sh: <https://gist.github.com/jbush001/a0fbe1e5f5eb8068a4667439af728a53>

gauravjain14, 2020-03-30:
Thanks for sharing the gist.

After using make, I was able to pass through the `./scripts/setup_tools.sh` step.
However, I seem to have hit another issue when I do `make` after doing `cmake .`

```
%Error: Internal Error: NyuziProcessor/hardware/testbench/soc_tb.sv:377: ../V3LinkDot.cpp:2222: Couldn't resolve inlined scope 'unnamedblk1' in: unnamedblk1
                                                                                                   : ... In instance soc_tb
                int'(nyuzi.l2_cache.l2_cache_read_stage.sram_l2_data.data[{way, set}] >> ((CACHE_LINE_WORDS - 1 - line_offset) * 32));
```

This seems to be related to verilator, but when I wrote a testbench to verify my installation of verilator that seemed to work fine.

jbush001, 2020-03-30:
Sorry this is having so many problems. I haven't seen this before either. Can you double check the version with this command?

   verilator --version

gauravjain14, 2020-03-30:
Thanks for actively responding. 
Output of `verilator --version`

`Verilator 4.030 2020-03-08 rev v4.030-4-ge69c380`

On other lines - is there a docker image for Nyuzi?

jbush001, 2020-03-30:
It appears you have a newer version of Verilator than is checked in (That is 4.012). Perhaps that is being installed by the package manager. It appears something broke with the latest version. I will need to debug that. I believe this could be fixed by uninstalling the system version and re-installing the version from the tree.

However, there is a docker package for this that contains Verilator and the compiler (it's used during CI testing), which would probably save some trouble. It is here: <https://hub.docker.com/r/jeffbush001/nyuzi-build>. Let me know if you have any problems with it.

gauravjain14, 2020-03-30:
Thanks for the pointers. 

When I let setup_tools.sh install verilator, it installs the version 4.03 (which I mentioned above). 

So I installed 4.012 from the verilator repository and then tried doing `make` again. Good News. Everything worked and I was able to run a couple of apps as well as able to pass all the tests by running `make tests`.

Thanks a lot for helping through this process. I can now go ahead and close the issue?

jbush001, 2020-04-04:
I've put in a workaround for the build issue, and I will open new issues to investigate the problems that were found here. Thank you for your help debugging this!

closed 2020-04-04 by jbush001