jbush001/NyuziProcessor #196: Suggestions

uis246, 2020-11-18:
1) Add IRC channel to README
2) What if add some per-thread single-instruction buffer. Will pipeline can stall one cycle instead two. E.g. when register dependency.

jbush001, 2020-11-23:
1. Yeah, I've thought about adding an IRC channel, but I only have a laptop, so I'd miss messages when offline. I've looked at some other chat services, but haven't found one that works great yet.
2. There is an instruction buffer for each thread that can buffer up to 8 instructions (https://github.com/jbush001/NyuziProcessor/blob/31f74e43a785fa66d244f1e9744ca0ca9b8a1fad/hardware/core/thread_select_stage.sv#L123). Or were you proposing putting a FIFO somewhere else?

uis246, 2025-02-12:
1. Matrix