jbush001/NyuziProcessor #199: Khronos ML summit

pboudier, 2021-09-21:
Hello,

I am the chair of a working group at Khronos developing standards for graphics and compute hardware accelerators (as well as a representative for Nvidia in those groups).

We are organizing a summit to gather feedback from the ecosystem, and influence the design of the next round of improvements, and I thought that you may be interested in being represented:
https://www.khronos.org/events/2021-invitation-to-the-khronos-machine-learning-summit

This summit is IP free, and will let you present your project and your needs for improvements in the ML ecosystem, as well as hearing from other companies.

regards,

Pierre Boudier
Software architect at Nvidia
chair of the Machine Learning TSG at Khronos

jbush001, 2022-05-01:
Apologies, I missed this one! Thank you for the invite, but it appears I'm a bit late. :)

closed 2022-05-01 by jbush001

pboudier, 2022-05-01:
If you are interested, khronos will have a permanent “machine learning forum” for recurring interactions between hardware and software vendors (multiple APIs). 

Pierre B.