jbush001/NyuziProcessor #201: run_fpga command does not work properly for me

ergintr, 2021-12-24:
Hey sir,
I synthesized the project from terminal and programmed it via quartus gui(could not do it via terminal), after that when I try ./run_fpga for different tests I just get yellow and white colors with no motion. The tests I have done are : 
* NyuziProcessor/software/apps/colorbars/run_fpga : I just get a full yellow screen on monitor
* NyuziProcessor/software/apps/rotozoom/run_fpga : I get the steady image below with no motion
[image]

*NyuziProcessor/software/apps/plasma/run_fpga : I get the steady image below with no motion 
[image]

Do you have any idea about this error ? 
Thank you.

jbush001, 2021-12-24:
When running in the testbench, the VGA signals aren't going to show anything interesting. That module is usually disabled, and, when it is not, the simulation doesn't run long enough to display interesting data, as will spend many cycles in the vblank (that module exists just to debug the memory transactions). 

As for the other issues: these are very strange. To start out, have you made any changes to the code (could you do a quick 'git diff' to check)? Which version of the Quartus tools are you using?

ergintr, 2021-12-28:
Thanks for reply, 
I use QUARTUS Lite 20.1 version. I have made the following changes :
* I have DE10-Standard fpga(https://www.terasic.com.tw/cgi-bin/page/archive.pl?Language=English&CategoryNo=165&No=1081) so I changed de2_115.qsf file for correct pin assigments of DE-10 standard.
* My fpga board has 16 bit SDRAM data width, so I changed sdram_controller.sv line :30 parameter DATA_WIDTH from 32 to 16.
* DE-10 standard has an HPS uart interface, so for uart communication I used GPIO of fpga board with an USB to serial module of FTDI(https://m.dx.com/p/mini-bluetooth-xbee-ft232rl-usb-to-serial-adapter-module-2079844).

jbush001, 2021-12-28:
Ah, thanks.
1. The graphics seem to only be yellow and white, which suggests the R and G lines aren't connected properly (stuck at 1), with only the B line connected. I'd suggest checking the pin configuration.
2. I don't believe the current SDRAM controller will work correctly if configured for 16 bits, as it doesn't handle the bus width and SDRAM width being different. This was a quick and dirty controller to bring up on my board, so it was hard coded as such. I'm a bit surprised this configuration works at all. If you can find another SDRAM module with an AXI interface (for example, I think Quartus may have one in their library), that might be easier, otherwise this would need a bit more logic to latch the upper and lower half words before putting them into the load_fifo (and likewise for stores).

jbush001, 2022-10-20:
Closing, as this is an unsupported configuration.

closed 2022-10-20 by jbush001

ergintr, 2022-10-20:
Thank you, for your help Sir.

jbush001, 2022-10-20:
Sorry there wasn't a better answer.