jbush001/NyuziProcessor #203: make error

Be997398715, 2022-09-08:
Hi, this is a great work, I just want to learn it, while I find these make error in my make process, examples:
1. %Error-UNSUPPORTED: /mnt/g/FPGA_ZYNQ_IC_Learn/IC/learn_project/GPU/NyuziProcessor/hardware/core/writeback_stage.sv:370:39: Unsupported: static cast to 'logic[15:0]' from 'logic[15:0]' : ... In instance soc_tb.nyuzi.core_gen[0].core.writeback_stage
code: writeback_value_nxt = vector_t'(mcycle_vcompare_result);
2. %Error-UNSUPPORTED: /mnt/g/FPGA_ZYNQ_IC_Learn/IC/learn_project/GPU/NyuziProcessor/hardware/core/writeback_stage.sv:400:39: Unsupported: static cast to 'logic[15:0]' from 'logic[31:0]'
                                                                                                                         : ... In instance soc_tb.nyuzi.core_gen[0].core.writeback_stage
code: writeback_value_nxt = vector_t'(ix_instruction.pc + 32'd4);
3. %Error-UNSUPPORTED: /mnt/g/FPGA_ZYNQ_IC_Learn/IC/learn_project/GPU/NyuziProcessor/hardware/core/writeback_stage.sv:402:39: Unsupported: static cast to 'logic[15:0]' from 'logic[15:0]'
                                                                                                                         : ... In instance soc_tb.nyuzi.core_gen[0].core.writeback_stage
code: writeback_value_nxt = vector_t'(scycle_vcompare_result);

What's wrong? And how I fix it?

jbush001, 2022-09-10:
It appears you are using the Xilinx tools here and have created your own build files (rather than the Altera/Quartus configuration that is in the repo). The Xilinx tools handle some cases in the language differently and apparently are more picky than the Quartus tools in this one. Specifically, the problem here seems to be that the Quartus tools will implicitly zero pad an assignment if the sizes aren't the same, whereas the Xilinx tools won't. In this case, the comparison result is 16 bits, but the writeback bus is 512 bits wide.

I don't have the Xilinx tools installed to test this, but I can suggest that you could pad the values out explicitly, something like:

```
                writeback_value_nxt = vector_t'({496'd0, mcycle_vcompare_result});
```

or perhaps assign only the low bits, i.e.:

```
                writeback_value_nxt[0] = {16'd0, mcycle_vcompare_result };
```

Be997398715, 2022-09-11:
> It appears you are using the Xilinx tools here and have created your own build files (rather than the Altera/Quartus configuration that is in the repo). The Xilinx tools handle some cases in the [...]

This is the process when I build this project. My settings are default. And I just try your method but no useful for this error like this below.
[image]
[image]
[image]

jbush001, 2022-09-11:
Okay, I was thrown because this was in a directory called 'FPGA_ZYNQ_IC_Learn'. It looks like maybe this is failing in Verilator? Can you type the following on the command line?

```
verilator --version
```

Be997398715, 2022-09-11:
> 

sry I just forgot telling you my environment, I work in the Ubuntu18.04 in WSL. The 'FPGA_ZYNQ_IC_Learn' directory is just a name which mounting on the disk. And the verilator version like below:
root@DESKTOP-355QV49:/mnt/g/FPGA_ZYNQ_IC_Learn/IC/learn_project/GPU/NyuziProcessor# verilator --version
Verilator 4.210 2021-07-07 rev v4.210

jbush001, 2022-09-11:
Thanks, that's helpful. The issue appears to be that you have a newer version of Verilator installed on your system than this was originally built with, and the latest version is stricter about how it handles casting. I have modified the Verilog to do this more cleanly, and run unit tests against the old Verilator to ensure it didn't introduce regressions, but haven't been able to test against the new version of Verilator, so no guarantees. Please sync to the latest revision and try again.

(the setup.sh does attempt to install a compatible version of Verilator, but, if there is another version installed, that can take precedence. Another potential remedy, if my patch above doesn't work. would be to uninstall the other system version and rerun setup.sh to install the older version).

I hope that helps.

jbush001, 2022-10-20:
Closing for now, as this should address the issue, but feel free to reopen if it is still a problem.

closed 2022-10-20 by jbush001