jbush001/NyuziProcessor #204: setup_tools.sh failure

LGarnet, 2022-10-19:
I have all the required packages and I've tried the python3 fix described in the readme, but upon running the script I get the following error:

```
CMake Error at /usr/share/cmake/Modules/ExternalProject.cmake:2471:EVAL:2 (add_custom_command):
  Error evaluating generator expression:

    $<TARGET_FILE:llvm-ranlib>

  Target "llvm-ranlib" is not an executable or library.
Call Stack (most recent call first):
  /usr/share/cmake/Modules/ExternalProject.cmake:2471 (cmake_language)
  cmake/modules/LLVMExternalProjectUtils.cmake:239 (ExternalProject_Add_Step)
  runtimes/CMakeLists.txt:216 (llvm_ExternalProject_Add)
  runtimes/CMakeLists.txt:275 (builtin_default_target)

CMake Error at /usr/share/cmake/Modules/ExternalProject.cmake:2471:EVAL:2 (add_custom_command):
  Error evaluating generator expression:

    $<TARGET_FILE:llvm-strip>

  Target "llvm-strip" is not an executable or library.
Call Stack (most recent call first):
  /usr/share/cmake/Modules/ExternalProject.cmake:2471 (cmake_language)
  cmake/modules/LLVMExternalProjectUtils.cmake:239 (ExternalProject_Add_Step)
  runtimes/CMakeLists.txt:216 (llvm_ExternalProject_Add)
  runtimes/CMakeLists.txt:275 (builtin_default_target)

CMake Error at /usr/share/cmake/Modules/ExternalProject.cmake:2471:EVAL:2 (add_custom_command):
  Error evaluating generator expression:

    $<TARGET_FILE:llvm-ranlib>

  Target "llvm-ranlib" is not an executable or library.
Call Stack (most recent call first):
  /usr/share/cmake/Modules/ExternalProject.cmake:2471 (cmake_language)
  cmake/modules/LLVMExternalProjectUtils.cmake:239 (ExternalProject_Add_Step)
  runtimes/CMakeLists.txt:216 (llvm_ExternalProject_Add)
  runtimes/CMakeLists.txt:275 (builtin_default_target)

CMake Error at /usr/share/cmake/Modules/ExternalProject.cmake:2471:EVAL:2 (add_custom_command):
  Error evaluating generator expression:

    $<TARGET_FILE:llvm-strip>

  Target "llvm-strip" is not an executable or library.
Call Stack (most recent call first):
  /usr/share/cmake/Modules/ExternalProject.cmake:2471 (cmake_language)
  cmake/modules/LLVMExternalProjectUtils.cmake:239 (ExternalProject_Add_Step)
  runtimes/CMakeLists.txt:216 (llvm_ExternalProject_Add)
  runtimes/CMakeLists.txt:275 (builtin_default_target)
  ```

LGarnet, 2022-10-19:
this is on arch linux

jbush001, 2022-10-20:
Thanks! It looks like there are multiple failures in the log. Can you do `cmake --version`?

LGarnet, 2022-10-20:
cmake version 3.24.1

jbush001, 2022-10-20:
I'm afraid I'm a bit stumped. It appears a number of things are failing.

In the interim, there is a Docker image with the tool chain here: https://hub.docker.com/r/jeffbush001/nyuzi-build. The CI script demonstrates how to use it: https://github.com/jbush001/NyuziProcessor/blob/master/.github/workflows/ci.yml.

LGarnet, 2022-10-20:
ah i see, thank you

jbush001, 2022-12-03:
This is actually an issue in https://github.com/jbush001/NyuziToolchain.

loonggi, 2023-10-15:
I encountered the same issue and fixed it by changing cmake from version 3.26.3 to 3.16.3.

jbush001, 2026-10-30:
Fixed in https://github.com/jbush001/NyuziToolchain/commit/4f2edda2c54eb762d2f8ee28a29dc838bf49ffd8

closed 2026-10-30 by jbush001