jbush001/NyuziProcessor #205: ./scripts/setup_tools.sh error on MAC M1

lgdu, 2022-11-02:
Hi,
I got these errors when i run ./scripts/setup_tools.sh on MAC M1 Platform.
```
CMake Error at /opt/homebrew/Cellar/cmake/3.24.2/share/cmake/Modules/ExternalProject.cmake:2474:EVAL:2 (add_custom_command):
  Error evaluating generator expression:

    $<TARGET_FILE:llvm-ranlib>

  Target "llvm-ranlib" is not an executable or library.
Call Stack (most recent call first):
  /opt/homebrew/Cellar/cmake/3.24.2/share/cmake/Modules/ExternalProject.cmake:2474 (cmake_language)
  cmake/modules/LLVMExternalProjectUtils.cmake:239 (ExternalProject_Add_Step)
  runtimes/CMakeLists.txt:216 (llvm_ExternalProject_Add)
  runtimes/CMakeLists.txt:275 (builtin_default_target)

CMake Error at /opt/homebrew/Cellar/cmake/3.24.2/share/cmake/Modules/ExternalProject.cmake:2474:EVAL:2 (add_custom_command):
  Error evaluating generator expression:

    $<TARGET_FILE:llvm-strip>

  Target "llvm-strip" is not an executable or library.
Call Stack (most recent call first):
  /opt/homebrew/Cellar/cmake/3.24.2/share/cmake/Modules/ExternalProject.cmake:2474 (cmake_language)
  cmake/modules/LLVMExternalProjectUtils.cmake:239 (ExternalProject_Add_Step)
  runtimes/CMakeLists.txt:216 (llvm_ExternalProject_Add)
  runtimes/CMakeLists.txt:275 (builtin_default_target)

CMake Error at /opt/homebrew/Cellar/cmake/3.24.2/share/cmake/Modules/ExternalProject.cmake:2474:EVAL:2 (add_custom_command):
  Error evaluating generator expression:

    $<TARGET_FILE:llvm-ranlib>

  Target "llvm-ranlib" is not an executable or library.
Call Stack (most recent call first):
  /opt/homebrew/Cellar/cmake/3.24.2/share/cmake/Modules/ExternalProject.cmake:2474 (cmake_language)
  cmake/modules/LLVMExternalProjectUtils.cmake:239 (ExternalProject_Add_Step)
  runtimes/CMakeLists.txt:216 (llvm_ExternalProject_Add)
  runtimes/CMakeLists.txt:275 (builtin_default_target)

CMake Error at /opt/homebrew/Cellar/cmake/3.24.2/share/cmake/Modules/ExternalProject.cmake:2474:EVAL:2 (add_custom_command):
  Error evaluating generator expression:

    $<TARGET_FILE:llvm-strip>

  Target "llvm-strip" is not an executable or library.
Call Stack (most recent call first):
  /opt/homebrew/Cellar/cmake/3.24.2/share/cmake/Modules/ExternalProject.cmake:2474 (cmake_language)
  cmake/modules/LLVMExternalProjectUtils.cmake:239 (ExternalProject_Add_Step)
  runtimes/CMakeLists.txt:216 (llvm_ExternalProject_Add)
  runtimes/CMakeLists.txt:275 (builtin_default_target)

```

jbush001, 2022-11-04:
Looks like the same issue as #204, except that was on Linux and this is on Mac. I'm guessing perhaps an incompatible change in a newer version of cmake or some other tool.

jbush001, 2022-12-03:
duplicate of #204

closed 2022-12-03 by jbush001